loathingKernel/ariadne-bootloader #14: Sketches will not upload to Mega

Nekraus, 2016-06-20:
Hello! 
I have a arduino Mega 2560 ans ITHEAD shield with W5100. I use Nick Gammon's guide to compile 
bootloader. I changed the  bit lock (0x0F) but the program indicates 0xCF. After the debug version send 
`Main: Ariadne for Arduino Mega2560, Version 0.5
 Net: Built-in settings
 Net: 	Address: 0xC0.0xA8.0x01.0x80
 Net: 	 Subnet: 0xFF.0xFF.0xFF.0x00
 Net: 	Gateway: 0xC0.0xA8.0x01.0xFE
 Net: 	    MAC: 0xDE.0xAD.0xBE.0xEF.0xFE.0xED
 Net: Network init done
Tftp: Init socket to port 0x0045`
but I can't flash any sketches via serial or tftp. I can't ping the arduino too.

LoathingKernel, 2016-06-21:
LockBit values 0xCF and 0x0F have equal meaning because bits 6 and 7 are not used in atmega2560. Being unable to upload with serial and not being able to ping the shield, probably means that the bootloader didn't initialize the shield properly, thus getting stuck while polling for an incoming connection to the shield.

I assume, since you followed Nick Gammon's guide, that you used an Uno to flash the new bootloader. I remember people having issues with that method, but I have to look through my backlog to see what was the problem and if there was a solution as it has been some time.

Nekraus, 2016-06-29:
hello! I used the olimex-mk2 but i have the same problem. No ping, timeout serial. Is it my shield which has a problem ?

Nekraus, 2016-06-29:
ok, I solved the problem. The shield should not be plug on the Mega. But now I would use the pin53 as SS, It is possible ?

LoathingKernel, 2016-06-29:
Yes, but you have to modify the code in the spi.h file and compile the booloader from the start.

titusece, 2016-12-23:
Hi Nekraus,
I'm facing similar issue like yours.
https://github.com/LoathingKernel/Ariadne-Bootloader/issues/15

I've also tried to flash the bootloader without shield on Mega board.
Any help please ?