loathingKernel/ariadne-bootloader #15: avrdude: stk500v2_ReceiveMessage(): timeout

titusece, 2016-12-23:
Hi,

I have a Arduino UNO (Arduino as ISP) , a Arduino MEGA2560 (to be programmed with Ariadne bootloader) and W5100 Ethernet shield.

As you have mentioned, I've followed the below link to program the Mega board with Ariadne-Bootloader(.h) using UNO(Arduino as ISP).
http://www.gammon.com.au/forum/?id=11635

Able to program the default Mega bootloader code using gammon's url.

I have converted "ariadne_atmega2560_w5100.hex" bootloader hex file to "ariadne_atmega2560_w5100.h" using MUSHClient and flashed the code into Mega2560 board.

Now I have tried to upload the blink code using USB cable (as usual) but I have got the below error.

Can you please help me on this issue ?

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM16
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
An error occurred while uploading the sketch
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

Regards,
Titus S.

titusece, 2016-12-23:
Moreover I have not connected the Ethernet Shield W5100 on Mega board while bootloader programming and serial uploading sketch.
Is that reason I'm getting timeout error ?

titusece, 2016-12-23:
In addition to that, after flashed bootloader, I've not seen Ethernet LED blinks (no activity).
What could be the problem ?

Nekraus, 2016-12-23:
Hello!
yeah you have to connect the ethernet shield but be carefull you must
connect the shield as explain here
https://www.arduino.cc/en/Reference/Ethernet
And it didn't work when the pins 11 to 13 was plugged on the arduino. I
bended the pins

sorry my english, it is not my native language

have a nice day !

:

titusece, 2016-12-23:
Thank you so much Nekraus!
Let me try it out :) :+1:

titusece, 2016-12-23:
Hi Nekraus,
I've programmed the bootloader to Mega board without shield on it.
After flashed the bootloader code, I've connected shield and checking the ethernet LED activity and tried to upload the code via USB interface on Mega board (which has the new ariadne bootloader)

Can you please provide your bootloader .h file of Mega2560 board ?

Thanks  for support.

titusece, 2016-12-23:
I've followed the below link for Arduino UNO(programmer) to Arduino MEGA(to be programmed with TFTP bootloader)
https://1.bp.blogspot.com/-369p0vA9eSo/V1NK1rlGftI/AAAAAAAATZI/QyG2WTfzIIc58ZLfBgdYvY4YqGVBFmYhwCLcB/s1600/Uno%2BArduino%2BISP%2Bto%2Bprogram%2BMega.png

Able to flash and boot the default bootloader from Gamma. (bootloader_atmega2560_v2.h)