loathingKernel/ariadne-bootloader #18: Bootloader burned successfully, Failed to upload bin via TFTP.

LoathingKernel, 2017-05-10:
_From @whitesn on August 18, 2016 6:26_

Hello, so I am able to burn the bootloader and upload the "ReadNetworkSettings" sketch using Serial USB Comm.

I am also able to ping the default IP when the Arduino is powered and connected (192.168.1.128)

Then I tried to upload the sample "blink" ( utilities\tests\blink\atmega328_blink9.bin ) by resetting the Arduino Board (LED13 now blinks quickly), then type "tftp -i 192.168.1.128 PUT atmega328_blink9.bin" (I already cd'd to the folder). After some times, it says "Connect request failed".

Previously, after uploading the sample "ReadNetworkSetting", the Arduino will be able to run the sketch after about 10 seconds after reset (the upload period).

But if I did attempt to TFTP the sketch (atmega328_blink9.bin), my Arduino is now not able to load the previous sketch and will keep blinking quickly (even after resetting, and waiting for 10 seconds). The previously uploaded sketch is now gone, but I am still able to upload sketches via serial. What should I do?.. (Windows 10 64bit, Arduino 1.6.7)

_Copied from original issue: codebendercc/Ariadne-Bootloader#34_

LoathingKernel, 2017-05-10:
_From @FunnyPaul on September 5, 2016 15:2_

Hi, I have also had problems with the TFTP tool of Windows. Try it once [http://tftpd32.jounin.net](http://tftpd32.jounin.net)  with the tool. And adjust or turn off the firewall.

LoathingKernel, 2017-05-10:
_From @whitesn on September 14, 2016 1:55_

Thanks @FunnyPaul , I'll get back after trying this out later.

OrangePledgeFanatic, 2025-03-30:
I am having pretty much the same issue, I am able to flash the bootloader, I am able to upload via serial with the bootloader, but windows tftp and tftpd64 don't work. The windows client comes back with connection request failed and tftpd64 is stuck at block 0. I am using the default setting and I am able to ping the arduino and shield. I am using a W5100 based shield.

OrangePledgeFanatic, 2025-03-30:
Nevermind, I had to add tftp.exe to the windows firewall, I thought installing the service would do that, and wasn't sure where it was. It is in system32 in case anyone is wondering.