loathingKernel/ariadne-bootloader #24: TFTP Fail when using ATmega1284 with WizNet5200

Borkman, 2017-05-17:
Hi everyone.
I have built a modular board to test different combinations of ATmega and Wiznet .

Trying these combinations I have found an issue with TFTP transfers when trying to send the .bin to an ATmega1284P + WizNet5200 board. While it pings correctly, TFTP is timeout
This is the combination I would like to use, and the only one with this issue. which is not present when using other combinations of ATmega and WizNet. I've tried it successfully with the ATmega1284P + WixNet5100, or with the ATmega328 + Wiznet5200 

I've tried with TFTP64 to discard windows TFTP client.
Any further ideas?

loathingKernel, 2017-05-17:
The only combinations tested by me are ATmega328 and ATmega2560 with W5100. I don't have any other hardware to test so I don't have first hand experience. I don't know if they work, but you could try one of the bootloaders marked as DEBUG and post the serial log here when you try to upload throught TFTP.

Borkman, 2017-05-25:
Hello again.

I tested a Debug version and it works, but then I realized that when I remove the USB2Serial interface it stopped working.

Then I went back to the "non debug" version again, and has the very same behaviour: TFTP works only when the USB2Serial interface is connected, but stops when disconnecting it.

It seems like something in the code is calling the serial port and when it doesn't reply it doesn't work.
I guess that shouldn't be happening on the non debug version.
If anyone could point me to where in the code is that reference to the serial interface, I can edit it and recompile, but I wasn't able to find it yet.

Borkman, 2017-07-03:
Hello again. No luck yet. Anyone?