loathingKernel/ariadne-bootloader #27: Problem with uploading some scetches

puhycz, 2017-09-04:
Hello, I use your ariadne bootloader with arduino mega 2560 and standard ethernet shield W5100. Now I have problem with upload this project. When I upload by USB cable standard from arduino IDE, it´s ok and function, but by TFTP it is not function - TFTP confirm transmission as success, but after uploads there is only LED still flashing. Smaller project is no problem upload by TFTP an your ariadne bootloader. Do you have some ideas why ?

There is some screenshot and bin and ino files:
 
https://www.puhy.cz/download/arduino/topeni/swtopeni.ino.ino
https://www.puhy.cz/download/arduino/topeni/topeni.bin
https://www.puhy.cz/download/arduino/topeni/tftp.png

but after this LED is only blinking.When I upload by Arduino IDE and USB cable it is ok and function ... 

Many thanks,

Jan

puhycz, 2017-10-23:
Now I have spend much of time by testing.
When I can upload some bigger bin, it is problem. I use different mega 2560 and W5100 boards for exclusion hw problem.... and there is timeout too ...

Small sketches f.e. blink ... is updated succesfully ...

F.e. I tested now your "ariadne-bootloader/utilities/tests/bigprogram/" and it´s and there is timeout too...

Please help me. Many thanks,

 Jan

[image]

loathingKernel, 2017-10-23:
It is a known issue and a proper fix is not easy.  I have been (very slowly) rewriting the bootloader to fix such issues but time is not permitting me to progress faster. 

As a workaround you can try to increase the time of retries your client does when a timeout occurs.

puhycz, 2017-10-23:
Thanks for your reply. I tested TFTPD64 with settings timeout 1s and max retransmit 100 and it is the same...

Timeout while waiting ack block #64 ... but before it LED is flashing ... arduino is reseting i think...

Do you have some tip for any function bootloader please? I need flashing remotely arduino Mega with W5100 or it is possible connect ESP8266.

Many thanks.
Jan

loathingKernel, 2017-10-23:
Ariadne only supports W5100, W5300 and W5500, so ESP8266 won't work with it. Any bootloader based on Ariadne or TFTP-booloader probably suffers from the same issue as far as I know. You could try some of the other forks though if they advertise Mega2560 support.

puhycz, 2017-10-25:
Now I try it with linux tftp and it s the same (fail on block 64)

`received ACK <block=0>
sent DATA <block=1, 512 bytes>
received ACK <block=1>
sent DATA <block=2, 512 bytes>
received ACK <block=2>
sent DATA <block=3, 512 bytes>
received ACK <block=3>
sent DATA <block=4, 512 bytes>
received ACK <block=4>
sent DATA <block=5, 512 bytes>
received ACK <block=5>
sent DATA <block=6, 512 bytes>
received ACK <block=6>
sent DATA <block=7, 512 bytes>
received ACK <block=7>
sent DATA <block=8, 512 bytes>
received ACK <block=8>
sent DATA <block=9, 512 bytes>
received ACK <block=9>
sent DATA <block=10, 512 bytes>
received ACK <block=10>
sent DATA <block=11, 512 bytes>
received ACK <block=11>
sent DATA <block=12, 512 bytes>
received ACK <block=12>
sent DATA <block=13, 512 bytes>
received ACK <block=13>
sent DATA <block=14, 512 bytes>
received ACK <block=14>
sent DATA <block=15, 512 bytes>
received ACK <block=15>
sent DATA <block=16, 512 bytes>
received ACK <block=16>
sent DATA <block=17, 512 bytes>
received ACK <block=17>
sent DATA <block=18, 512 bytes>
received ACK <block=18>
sent DATA <block=19, 512 bytes>
received ACK <block=19>
sent DATA <block=20, 512 bytes>
received ACK <block=20>
sent DATA <block=21, 512 bytes>
received ACK <block=21>
sent DATA <block=22, 512 bytes>
received ACK <block=22>
sent DATA <block=23, 512 bytes>
received ACK <block=23>
sent DATA <block=24, 512 bytes>
received ACK <block=24>
sent DATA <block=25, 512 bytes>
received ACK <block=25>
sent DATA <block=26, 512 bytes>
received ACK <block=26>
sent DATA <block=27, 512 bytes>
received ACK <block=27>
sent DATA <block=28, 512 bytes>
received ACK <block=28>
sent DATA <block=29, 512 bytes>
received ACK <block=29>
sent DATA <block=30, 512 bytes>
received ACK <block=30>
sent DATA <block=31, 512 bytes>
received ACK <block=31>
sent DATA <block=32, 512 bytes>
received ACK <block=32>
sent DATA <block=33, 512 bytes>
received ACK <block=33>
sent DATA <block=34, 512 bytes>
received ACK <block=34>
sent DATA <block=35, 512 bytes>
received ACK <block=35>
sent DATA <block=36, 512 bytes>
received ACK <block=36>
sent DATA <block=37, 512 bytes>
received ACK <block=37>
sent DATA <block=38, 512 bytes>
received ACK <block=38>
sent DATA <block=39, 512 bytes>
received ACK <block=39>
sent DATA <block=40, 512 bytes>
received ACK <block=40>
sent DATA <block=41, 512 bytes>
received ACK <block=41>
sent DATA <block=42, 512 bytes>
received ACK <block=42>
sent DATA <block=43, 512 bytes>
received ACK <block=43>
sent DATA <block=44, 512 bytes>
received ACK <block=44>
sent DATA <block=45, 512 bytes>
received ACK <block=45>
sent DATA <block=46, 512 bytes>
received ACK <block=46>
sent DATA <block=47, 512 bytes>
received ACK <block=47>
sent DATA <block=48, 512 bytes>
received ACK <block=48>
sent DATA <block=49, 512 bytes>
received ACK <block=49>
sent DATA <block=50, 512 bytes>
received ACK <block=50>
sent DATA <block=51, 512 bytes>
received ACK <block=51>
sent DATA <block=52, 512 bytes>
received ACK <block=52>
sent DATA <block=53, 512 bytes>
received ACK <block=53>
sent DATA <block=54, 512 bytes>
received ACK <block=54>
sent DATA <block=55, 512 bytes>
received ACK <block=55>
sent DATA <block=56, 512 bytes>
received ACK <block=56>
sent DATA <block=57, 512 bytes>
received ACK <block=57>
sent DATA <block=58, 512 bytes>
received ACK <block=58>
sent DATA <block=59, 512 bytes>
received ACK <block=59>
sent DATA <block=60, 512 bytes>
received ACK <block=60>
sent DATA <block=61, 512 bytes>
received ACK <block=61>
sent DATA <block=62, 512 bytes>
received ACK <block=62>
sent DATA <block=63, 512 bytes>
received ACK <block=63>
sent DATA <block=64, 512 bytes>
sent DATA <block=64, 512 bytes>
sent DATA <block=64, 512 bytes>
sent DATA <block=64, 512 bytes>
sent DATA <block=64, 512 bytes>
Transfer timed out.
`

hagaigold, 2017-11-10:
I don't now if there is a difference, but with **W5500**, I was able to upload a 250k sketch to mega2560.
it was so slow, so I reduced the `#define TFTP_PACKET_DELAY` to 20 ms. [worth trying](https://github.com/loathingKernel/ariadne-bootloader/blob/cfd07b3598ed2e68ca79916d98befffb55136fa2/hardware/ariadne/bootloaders/ariadne/tftp.h#L53), I think..

>

loathingKernel, 2017-11-10:
With W5500 there is a possibility that you could remove the delay completely as it is. You don't encounter these issues with  it because W5500 increment its socket buffer pointers internally, unlike W5100 and W5200.

hagaigold, 2017-11-10:
in my tests i was able to go down to 5 ms with W5500.. I didn't try less that this.

puhycz, 2018-01-09:
So, I buy Arduino Ethernet Shield 2 (W5500), connect to Arduino Mega 2560. Ping to 192.168.1.128 is ok, but is not possible to upload some (small or big) file. LED still flashing ...

Is there some specialy diferences for ethernet shield 2?