loathingKernel/ariadne-bootloader #28: Bootloader problems

ChrisFredriksson, 2017-10-30:
Hey there,
First off, everyone who has worked with this in the past, big Thanks! :) Also to loathingKernel for your wonderful continuation!

I'm having a look at this bootloader as I would like to use it for programming a couple of units in a remote location.. Which this bootloader should be able to do, which is great :)

However, I'm having a few problems.

Which software are you using for editing and uploading the bootloader to the target?

- I've tried Arduino 1.8.3 which seems to be a no go. I've tried to put the "hardware" and "libraries" both under <User Documents folder>/arduino/ and also in the <AppData>/arduino path, but also in the "C:/program.../arduino" path but I just get errors that the board.txt file can't be read and similar. I can of course provide more errors when I'm told exactly where the files should be put, but as of right now, when I follow the Readme file, it just doesn't work.

- I've also tried Atmel Studio 7 which can compile the code, create an HEX file and upload it to the target. However, I still can't ping the device either on the default IP which ends in .128 or the IP I've set which should be .33

I've edited the net.h file with my network settings, like the following;
`/* Network settings */
#define IP_ADDR     192,168,1,33
#define SUB_MASK    255,255,255,0
#define GW_ADDR     192,168,1,1
#define MAC_ADDR	0x55,0xD7,0x53,0x00,0x01,0x0B`

But, I can't ping the device on this address. The Gateway address I suppose should be the real network gateway, such as the router, correct?

Also, how do I enable debugging in the source code, so that it outputs data to the Serial port, just like the "ariadne_debug2560_w5100.hex" file does when it is uploaded?

When I upload the bootloader file "ariadne_debug2560_w5100.hex" I do get debug output and the IP address is set to 192.168.1.128 - which by the way is matching my network and it is a free IP. I still can't ping the device though.

The device I'm using is actually an Controllino Maxi. This device is based off the Atmega2560 with a WizNet 5100 chip. The following are the pin connections which is hardwired and can't be changed in any way.

`Arduino 52 / PB1 - WizNet SCK
Arduino 51 / PB2 - WizNet MOSI
Arduino 50 / PB3 - WizNet MISO
Arduino xx / PJ3 - WizNet SS/CS`

I've thought that maybe the PJ3 should be configured in the spi.h file as the chipselect, but not sure if that does anything? I've tried changing both #define SS and #define ETH_SS to PINJ3, but no change in response from the unit on the network.

Something I've noticed is that the PINB7, defined as LED in spi.h, the led is not flashing or blinking, just solid on.

Also, I don't see any traffic on WireShark that could be the bootloader. But I guess thats only when inbound traffic is active. The bootloader isn't transmitting anything without an incomming call, right?

Pinout for the Controllino, Ethernet chip just right of center on the pinout sheet, below SD card connector..
http://controllino.biz/wp-content/uploads/2017/05/CONTROLLINO-MAXI-Pinout.pdf

I think that is all for the moment, would it possible to help me out in any way? :)

Many thanks in advance!

// Christopher

hagaigold, 2017-11-10:
>But, I can't ping the device on this address. The Gateway address I suppose should be the real network gateway, such as the router, correct?

can you ping a simple test sketch that just setup your ethernet device?
I would start here before moving to the bootloader.

> Also, how do I enable debugging in the source code, so that it outputs data to the Serial port, just like the "ariadne_debug2560_w5100.hex" file does when it is uploaded?

you can use the make file to compile a  _debug_ vresion.
e.g. 

- make debug2560_w5500 for debug version
- make atmega2560_w5500 with no debug.

or just see what the make file do and do the same in your code (hint: some symbols should be define for enable debuging).

ChrisFredriksson, 2017-12-19:
Hello Hagaigold,
Thank you for the response! I'm really sorry I haven't replied yet, I could come up with several reasons, but I'm sure you're not interested ;)

I can ping the device with other ethernet sketches yes, so the hardware should work properly so far. What I don't know however is if some other pins might be used to control the W5100 on the hardware I use - Controllino Maxi.

[image]

Should I be able to see something, with for example Wireshark, on the network when the bootloader works and connects to the network?

I tried, when I wrote my original message, to define variables to enable debugging, but I just ran into errors and then I lost hope in my abilities or something, I haven't tried since ;)

I'm not so used with make files, but I guess I would need the "make" program to be able to compile something out of the make-file, correct? I'm on Windows 10, so I need to figure out how to install that first as it seems that it didn't install with either AVR Studio or the Arduino IDE.

Really appreciate your reply and help! I'm heading for Christmas vacation tomorrow and I hope to have some time over to get back into this project and see if I can get it to work :D

hagaigold, 2017-12-20:
> I've tried changing both #define SS and #define ETH_SS to PINJ3

If I understand the scheme of your hardware, this setup seems wrong to me.
While it look like you probably need to change ETH_SS to PINJ3, SS should stay as it is `#define SS      PINB0`, otherwise, the SPI on the atmega2560 will not be configure correctly (PINB0 must set HIGH and output for the SPI to function).
also make sure the SD_SS pin not interfere with your hardware, because it also part of the SPI init.

see the code from spiInit:

```
	/** Set SPI pins high */
	SPI_PORT = _BV(SCK) | _BV(MISO) | _BV(MOSI) | _BV(SS);
	/** Set SPI pins as output */
	SPI_DDR = _BV(SCK) | _BV(MOSI) | _BV(SS);

	#if (ETH_SS != SS)
	/** Initialize extra SS pin used in some boards (mega) */
	/** Set ethernet SS high */
	ETH_PORT |= _BV(ETH_SS);
	/** Set ethernet SS as output */
	ETH_DDR |= _BV(ETH_SS);
	#endif

	/** Disable SD card */
	/** Set SD SS pin high */
	SD_PORT |= _BV(SD_SS);
	/** Set SD SS pin as output */
	SD_DDR |= _BV(SD_SS);
```

but, you don't need take my advice here about the pins setup.
if you really able to run a sketch with your hardware and ping it, this tell me you have a working version of Ethernet library, probably provide by Controllino, so juts look how it is setup there, and do the same in the bootloader.

----

It is no so hard to setup a basic environment on windows box, just get the following 3 tools:
- avr8-gnu-toolchain
- avrdude-6.3
- Make for Windows (from gnuwin32)

they all freely available  (just google it).
avr-gnu & avrdude also come with Arduino IDE, and buried somewhere in the installed folder, but in my setup I just downloaded them as stand alone, so It should work for you as well.

after you set them up, let's say under C:\Tools\ folder, create a batch script like below:
** adjust the path as required

```
echo off

SET gccroot=C:\tools\avr8-gnu-toolchain\bin
SET avrdude=C:\Tools\avrdude-6.3-mingw32
SET make=C:\tools\utils\bin

echo Adding gcc path - %gccroot%
echo Adding avrdude path - %avrdude%
echo Adding make path - %make%

set path=%path%;%gccroot%;%avrdude%;%make%
```

- open command prompt in ariadne-bootloader source folder
  ( `....\Ariadne-Bootloader\hardware\ariadne\bootloaders\ariadne`)
- run the batch file to set up the paths
- and then make your code by `make debug2560_w5100`, or, `make atmega2560_w5100 `

you should find now in the same folder a compiled hex files.

if you successfully compiled the code, now it time to start explore & introduce changes :)
hope this will help you going forward..

Happy Christmas

jkroeze, 2018-04-22:
Hi,

I also try to flash a Controllino Maxi with this bootloader. Did you find a solution to your problem?

Thanks in advance!

ChrisFredriksson, 2018-04-23:
Hey **jkroeze**,
Sorry, I haven't actually! **hagaigold** did write a nice guide, but I've just pushed this one away for some bit to get some distance.. even though I don't need the distance, I need to fix it ;) hehe

I'm gonna try to see if I can get this to work this week, I have a couple of other things I need to fix also, but I'm going to prioritize this now because I really need it :)

How far have you come **jkroeze**?

ChrisFredriksson, 2018-04-24:
Hey jkroeze,
Alright, so the modified files, the spi.c and spi.h are from the Controllino files that are/should be included alongside the Controllino library - when you click include library->Controllino?

And then both the per1234 and the loathingKernel works with ETH upload?

While loathingKernel version does not work with serial upload?

Am I correct so far?

I think I read a thread somewhere here earlier about increasing the packet size to allow Mega2560 to be serially programmed as well, or perhaps that was for the ETH programming.. I have to look that up, but perhaps you know or someone else know of this already?

Now I got even more enthusiastic about programming the Controllino with the bootloader, I think I will have some time over tomorrow wednesday to fix with this..

Not saying that I "knew" it or anything, but I think I was on the track with the pins earlier that it might be the wrong pin used for the selectpin.. But I guess you beat me to fixing that :D I wouldn't have found that so fast either, I guess ;)

So, great work!

ChrisFredriksson, 2018-07-09:
Hey again,
Finally got some time to test and try to figure out how this is supposed to work :)

Using the Arduino IDE, I can install the library/board thingies through the Board Manager without problems. However, programming my Controllino MAXI with an ATMEL-ICE via ISP does not work.

I get the following error;
`avrdude: usbdev_send(): wrote -22 out of 7 bytes, err = 
avrdude: jtag3_send(): failed to send command to serial port
avrdude: failed to sync with the JTAGICE3 in ISP mode
Error while burning bootloader.`
And it seems that more people have the same problem and there is no direct solution..

So, headed over to Atmel Studio 7 which has no problem programming the Controllino though the ATMEL ICE and over ISP. I can successfully program the unit but I still get the "freeze" of the Controllino MAXI after startup.

I both tried the HEX file that you included in the ZIP file, and I've tried with the complete clean code as a new project - just replacing the spi.c and spi.h files - the ones you had updated.

I still get the same "freeze" of the Controllino MAXI unit.

With freeze I mean that D2/PG5, D8/PB4, D11/PB7 are lit (LED's connected to these on the Controllino MAXI). There seems to be no direct activity on the network port and nothing else seems to happen.

I can't seem to ping 192.168.1.128 either, my LAN is setup for 192.168.1.x - so the unit should be able to cozy around on the network. The computer/server I'm trying to talk to my Controllino with is a Linux Debian machine, with no restrictions when it comes to firewalls and such. The computer/server and the Controllino are the only ones sitting on the same LAN. The router used for this separate network is an Edgerouter, but the "firewall" part of the router should most certainly not block the tftp traffic within the LAN.

There are also no DHCP addresses dealt with, but I guess that's fine as the Ariadne bootloader by default wants 192.168.1.128 as an fixed IP.

My computer/server is using 192.168.1.5. I'm not sure if the "gateway" as found in the code of the Ariadne bootloader, should be the tftp computer/server? If so, why would it be default set to 192.168.1.1 - which in 99% of cases are a real gateway? aka. router or similar. I've tried changing the source to use 192.168.1.5 as the "gateway", but seems to be same result.

Also, what is needed by the computer/server? I should be able to use a tftp client on it to connect to the Ariadne bootloader, which acts as the tftp server - correct?
Trying the following;
`tftp
connect 192.168.1.128
put test.txt`
And after a while I get "Transfer timed out.", most probably because the Controllino is not properly connected to the network.

So I can't say that I believe my unit has network connection in the bootloader mode. While the unit does have network connection when loading another sketch/program into the Controllino - compiling my regular sketch in the Arduino IDE, taking the HEX file and programming the Controllino through AS7 with the Atmel ICE.

It's really interesting that even your HEX file, **jkroeze**, does not work on my Controllino MAXI.

I think I've got all the details on how my network is setup, how I programmed the unit and how I try to connect to it.

Perhaps I've missed something, compared to your setup **jkroeze**?

ChrisFredriksson, 2018-07-10:
Progress! Well.. not thanks to me I guess.. but I've noticed a difference!

I've got both the Atmel Studio compilation to get network access now with IP 192.168.1.128 when gateway is set both to 192.168.1.1 which is my router, and to 192.168.1.5 which is my computer/server. But I guess it really should be the router address that should be the gateway, correct?

However it doesn't matter which gateway address I specify, the Ariadne bootloader doesn't seem to answer at all.

Another thing I've noticed is that the Controllino when started isn't responding on ping until after 10 seconds. After this period it responds to ping like normal;
```
PING 192.168.1.128 (192.168.1.128) 56(84) bytes of data.
64 bytes from 192.168.1.128: icmp_seq=10 ttl=128 time=0.138 ms
64 bytes from 192.168.1.128: icmp_seq=11 ttl=128 time=0.120 ms
```

It also crossed my mind to check with you guys if my fuses are correct, if perhaps you guys have changed them on your projects?
```
Extended	0xFD
High		0xD0
Low		0xFF
Lockbit		0xFF (nothing set)
```

I tried enabling debug today and all I get out of it is, continuously;
```
ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
```
Using 115200 as baud and ASCII. When using HEX presentation instead, I get 0xFF

I've tried enabling debug for each and all of the following;
```
#define DEBUG_MAIN	1
#define DEBUG_SPI		1
#define DEBUG_NET	1
#define DEBUG_TFTP	1
```
Same thing happens for all, except when enabling it for main, the program also stops before the LED's are lit.

I have also, with the above defined the following to enable debug
```
#define ENABLE_DEBUG
```

I have also tried the following and I don't get the proper output on any terminal program;
```
puthex(0xAF);
putch(0xF0);
putch("X");
```

More or less random crap is coming out, sometimes its the same output but far from always.. Quite strange I guess..

The commands I use for tftp are the following, compiled into an .sh file on my Linux system
```
tftp 192.168.1.128 << 'EOF'
mode octet
verbose
trace
put test.bin
quit
EOF
```

With that I get the following;
```
root@bootloader:/# ./tftp.sh

tftp> Verbose mode on.
tftp> Packet tracing on.
tftp> putting test.bin to 192.168.1.128:test.bin [octet]
sent WRQ <file=test.bin, mode=octet>
sent WRQ <file=test.bin, mode=octet>
sent WRQ <file=test.bin, mode=octet>
sent WRQ <file=test.bin, mode=octet>
sent WRQ <file=test.bin, mode=octet>
Transfer timed out.
```

Finally, I guess that the LED - specified in spi.h (LED_DDR/LED_PORT/LED) should blink with the main loop in the file main.c through the use of updateLed(); function. Either it does blink but extremely fast so it looks like its always on, or it never blinks.

However, it does cut out every  ~6-8 seconds - when the WDT resets the unit as it should without a userprogram uploaded.

----------------------

I found where the program is during the course of the ~8 seconds before WDT resets the unit.
Apparently the program is running sockInit(*) in file tftp.c, it is stuck here completely.

I can find this out if I run this "modified" sockInit(*) function;
```
static void sockInit(uint16_t port) {
    DBG_TFTP(
        tracePGMlnTftp(mDebugTftp_SOCK); tracenum(port);
    )

    spiWriteReg(REG_S3_CR, S3_W_CB, CR_CLOSE);
    while (spiReadReg(REG_S3_CR, S3_R_CB)) {
        //wait for command to complete
    }

    do {
        // Write interrupt
        spiWriteReg(REG_S3_IR, S3_W_CB, 0xFF);
        // Write mode
        spiWriteReg(REG_S3_MR, S3_W_CB, MR_UDP);
        // Write TFTP Port
        spiWriteWord(REG_S3_PORT0, S3_W_CB, port);
        // Open Socket
        spiWriteReg(REG_S3_CR, S3_W_CB, CR_OPEN);
        while (spiReadReg(REG_S3_CR, S3_R_CB)) {
            //wait for command to complete
        }
        // Read Status
        if (spiReadReg(REG_S3_SR, S3_R_CB) != SOCK_UDP)
            // Close Socket if it wasn't initialized correctly
            spiWriteReg(REG_S3_CR, S3_W_CB, CR_CLOSE);

        LED_PORT ^= _BV(LED);
        _delay_ms(30);
        LED_PORT &= ~_BV(LED);
        _delay_ms(30);
        LED_PORT ^= _BV(LED);
        _delay_ms(30);
        LED_PORT &= ~_BV(LED);
        _delay_ms(30);
        LED_PORT ^= _BV(LED);
        _delay_ms(30);
        LED_PORT &= ~_BV(LED);
        _delay_ms(30);
        LED_PORT ^= _BV(LED);
        _delay_ms(30);
        LED_PORT &= ~_BV(LED);
        _delay_ms(30);

        // If socket correctly opened continue
    } while (spiReadReg(REG_S3_SR, S3_R_CB) != SOCK_UDP);
}
```
When I run this modified version the LED blinks rapidly until the WDT resets the unit.

This means that everything in the main loop in the file main.c is not run after this part;
```
	/* Initialize the UDP socket for tftp */

	DBG_MAIN_EX(tracePGMlnMain(mDebugMain_TFTP);)

	tftpInit();
```

Is this wrong or should it be like this??

ChrisFredriksson, 2018-07-10:
More progression! Starting to become a long monologue here ;)

I found that if I remove this part in file spi.c, from your file jkroeze
```
	RTC_PORT |= _BV(RTC_SS);
	/** Set RTC SS pin as output */
	RTC_DDR |= _BV(RTC_SS);
```

Or the following from the file spi.c, when using the loathingKernel project
```
	/** Disable SD card */
	/** Set SD SS pin high */
	SD_PORT |= _BV(SD_SS);
	/** Set SD SS pin as output */
	SD_DDR |= _BV(SD_SS);
```

The bootloader does not stop at the sockInit(*) function. It now runs on and responds directly on pings, however the upload does not seem to work still..

Another fantastic discovery is that the bootloader from;
[https://github.com/per1234/Ariadne-Bootloader](url)
Now works perfectly as well, even more so as it also accepts uploads and it does upload to the microcontroller via TFTP. However, for some reason the uploaded program does not start after the upload is completed, not even when resetting the Controllino. So I guess that the bootloader does not timeout and does not find the user program at position 0x0000.

Is there something special I need to do with the user program? It doesn't have to be a specific size either right?

As far as I can tell I should be able to upload an Arduino sketch (tried a regular Blink sketch) after converting it with avr-objcopy. I've used the following command to do so;
```
avr-objcopy -j .text -j .data -O binary Blink.ino.elf upload.bin
```

Any help on this would be, of course, much appreciated :) I'm so happy with the progress I've made so far and I got the unit to finally be "online" and responding at least. Now there is only a small step, as it feels like, to get to the finish line :)

ChrisFredriksson, 2018-07-12:
An update on the previous, just to show the TFTP communication

This is with the per1234 version of the Ariadne Bootloader, shouldn't there be something else after the part with "Sent 2436 bytes in 4.0 seconds [4872 bits/sec]" ? Like restarting, completed, starting user program or anything?
```
tftp> Verbose mode on.
tftp> Packet tracing on.
tftp> Connected to 192.168.1.128.
Mode: octet Verbose: on Tracing: on
Rexmt-interval: 5 seconds, Max-timeout: 25 seconds
tftp> putting upload.bin to 192.168.1.128:upload.bin [octet]
sent WRQ <file=upload.bin, mode=octet>
received ACK <block=0>
sent DATA <block=1, 512 bytes>
received ACK <block=1>
sent DATA <block=2, 512 bytes>
received ACK <block=2>
sent DATA <block=3, 512 bytes>
received ACK <block=3>
sent DATA <block=4, 512 bytes>
received ACK <block=4>
sent DATA <block=5, 388 bytes>
received ACK <block=5>
Sent 2436 bytes in 4.0 seconds [4872 bits/sec]
```

This is with the LoathingKernel version of the Ariadne Bootloader
```
tftp> Verbose mode on.
tftp> Packet tracing on.
tftp> Connected to 192.168.1.128.
Mode: octet Verbose: on Tracing: on
Rexmt-interval: 5 seconds, Max-timeout: 25 seconds
tftp> putting upload.bin to 192.168.1.128:upload.bin [octet]
sent WRQ <file=upload.bin, mode=octet>
sent WRQ <file=upload.bin, mode=octet>
sent WRQ <file=upload.bin, mode=octet>
sent WRQ <file=upload.bin, mode=octet>
sent WRQ <file=upload.bin, mode=octet>
Transfer timed out.
```

loathingKernel, 2018-07-12:
> This is with the per1234 version of the Ariadne Bootloader, shouldn't there be something else after the part with "Sent 2436 bytes in 4.0 seconds [4872 bits/sec]" ? Like restarting, completed, starting user program or anything?

No, the messages you see are from your tftp client. Ariadne doesn't sent anything back except for errors, so you shouldn't see anything other than that. 

AFAIK, @per1234's fork doesn't touch the bootloader code, other than adding atmega1284P support, it is in its entirety an older version of this repository with conveniences for arduino IDE. If one works for you and the other doesn't, start by looking at the commit history to see what has changed.

per1234, 2018-07-12:
I think loathingKernel is correct about the bootloader differences, though it's been so long I really don't remember.

Comparison of my fork to this repo:
https://github.com/loathingKernel/Ariadne-Bootloader/compare/master...per1234:ariadne
vice-versa:
https://github.com/per1234/Ariadne-Bootloader/compare/ariadne...loathingKernel:master

Another difference might be in the compilation of the bootloader. My hardware package provides .hex files, which were compiled with the version of avr-gcc included with the Arduino IDE at that time so maybe you're using those .hex files? I'm guessing you're compiling the bootloader in this repo yourself.

Another consideration is fuses. You should verify the fuses are the same in each case. Application not starting can definitely be caused by misconfigured fuses.

ChrisFredriksson, 2018-07-12:
Thanks for coming here and replying as well per1234! :)

Both the per1234 and the loathingKernel version is compiled in Atmel Studio 7, burned with an Atmel ICE through AS7. I wanted to get everything working directly through the Arduino IDE, and sure enough the download through board manager works and all that. However, for some reason (more people have the same problem without solution, as far as I know) the Atmel ICE does not work with the Arduino IDE, at least not if you have AS7 installed.

The fuses should be the same, no matter which project, but just to be sure I checked them and they are
```
Extended	0xFD
High		0xD0
Low		0xFF
Lockbit		0xFF (nothing set)
```

The bootloader program in both cases are starting and they get online so that I can ping them. Its just after the per1234 version has received the file it doesn't start up the uploaded binary file. loathingKernel version doesn't even respond to TFTP connection. So, I guess that this shouldn't be due to wrong fuses, but in that case you guys know more than me :)

Also, you're mentioning .hex files. Well if I burn one of the hex files, I think only the ones from the per1234 repo, I can even get the debug working. However, as the pinout is different in the spi.h file the W5100 isn't started so I don't get any network connectivity.

But when trying to enable debugging in either of the two repos, when compiling it in AS7 I only get crap out on the terminal. Also, if I define debug main, the controller locks up before the led lights up, and no network connectivity.
```
#define DEBUG_MAIN  1
```

Of course, I've tried both with and without defining;
```
#define ENABLE_DEBUG
```

If I could get the debug, or even the serial to work I could at least find if the loathingKernel version sees any incoming connection, or if the per1234 version validates the binary file and is ready for restart or whatever.

I've also put the following in all files where delay.h is included to see if that helped, as it could be a speed issue when the serial line isn't working properly, but without any positive result.
```
#define F_CPU 16000000UL
```

Furthermore I've tried to lower the baudrate but no difference there either.

I don't really know what else to test.. I guess my only bet now is to compare the repos and see if I can find something..

But at least it would be good to know if the per1234 version at least saves the binary correctly, or if its just bogus that comes in/out.

I don't have to do anything special with the binary file correct? I've tried the following two commands to create the binary file, after it has been compiled by the Arduino IDE, just a simple blink sketch for led on D0.
```
avr-objcopy -I ihex Blink.hex -O binary upload.bin
avr-objcopy -j .text -j .data -O binary Blink.ino.elf upload.bin
```

ChrisFredriksson, 2018-07-13:
Aha! I will have a go at it this weekend and see if that helps, didn't even think about that. I guess I thought that it was completely necessary only due to errors in upload or something.

But that sounds resonable and I now understand how it works, so shouldn't be too hard to disable it :)

Many thanks!

hagaigold, 2018-07-14:
I had many problem with the serial bootloader.. so I changed it to work like in Arduino project.
the fix was in private repository, now it can be found here: 55cdbc4a4e96d09dc6d4c2d516cf1ca52217404d

loathingKernel, 2018-07-15:
The two methods are supposed to be equivalent according to the datasheet, but if one works better than the other on certain cases, it should be brought back.

I am more interested as to why was the change in a private repository?

hagai-mego, 2018-07-15:
the difference, and where the problem coming from, is from the timing of the sending loop(while loop).
endchar: [update UART_DATA_REG] .. [while waiting]
putch: [while waiting] .. [update UART_DATA_REG]

in my testing, sometime the last byte was getting "lost": avrdude was never seen it, and emits errors.
probably some old avrdude  versions, didn't care so much or maybe knew better to wait..

hagaigold, 2018-07-15:
> I am more interested what you mean by this.
Comment been update, it is mean it was not pushed to the public repository.
but i am not clear how this relevant to the context of the problem.

hagaigold, 2018-07-15:
> The two methods are supposed to be equivalent

the difference, and where the problem coming from, is from the timing of the sending loop(while loop).
sendchar: [update UART_DATA_REG] .. [while waiting]
putch: [while waiting] .. [update UART_DATA_REG]

in my testing, sometime the last byte was getting "lost": avrdude was never seen it, and emits errors.
probably some old avrdude versions, didn't care so much or maybe knew better to wait..

loathingKernel, 2018-07-15:
> Comment been update, it is mean it was not pushed to the public repository.
but i am not clear how this relevant to the context of the problem.

As it was worded, it wasn't clear to me if by private you meant your local repository, or some other private repository.