loathingKernel/ariadne-bootloader #34: tftp issue with Netload and serial issue with Ariadne

redmuchroom, 2019-08-08:
Hi,

I did first a upload of the bootloader of Ariadne on a Atmega2560 (in fact, a controllino Mega). I did some modifications in spi.h, spi.c for pinout on the eth et rtc (instead of sd). 
(I uploaded with an arduinoISP with the http://www.gammon.com.au/forum/?id=11635 scketch)

I could make an upload through tftp and software was working fine but I got some issues to ping the atmega and to reupload through tftp. the device was unreachable... Also I got issue to upload the sketch on serial (via arduino IDE) with another computer (with mine it is still ok...?) : uploading is hanging...

then I tried with the Netload bootloader to check... There I can upload on serial. I can update the network data and I can ping the atmega... However I get a timeout when I try to upload through tftp (I disabled the firewall...) 

any idea?

thanks !

loathingKernel, 2019-08-08:
>  However I get a timeout when I try to upload through tftp (I disabled the firewall...)
NetLoad is an older fork of Ariadne, I don't remember if Atmega2560 was supported at the time.

> I could make an upload through tftp and software was working fine but I got some issues to ping the atmega and to reupload through tftp. the device was unreachable...
Do you setup the network inside the sketch you are uploading to the arduino?

> Also I got issue to upload the sketch on serial (via arduino IDE) with another computer (with mine it is still ok...?)
This is weird, possible driver issue?

redmuchroom, 2019-08-08:
thanks for the quick reply.

> NetLoad is an older fork of Ariadne, I don't remember if Atmega2560 was supported at the time.
so better to continue with Ariadne...
>Do you setup the network again inside the sketch you are uploading to the arduino?
No, I do it only once with the eeprom sketch...
>This is weird, possible driver issue?
agree, this weird :). I don't think this is a driver issue as it is working fine with the standard bootloader...

thanks

loathingKernel, 2019-08-08:
It has been some time since I last worked on this project, so things might have changed on the Arduino side. I can mostly offer suggestions, as I have no hardware available at the moment.

> No, I do it only once with the eeprom sketch...

If you are using the Ethernet library, it might be getting the IP through DHCP if you have not manually specified it or you are not reading it from EEPROM. If you are not using it, on paper WizNet should keep its configuration unless it is resetted during sketch boot (unlikely but keep it in mind in case I am missing something). In either case, specify the network configuration in the sketch too.

> I don't think this is a driver issue as it is working fine with the standard bootloader...

Ariadne requires its own board in the board selection menu to upload correctly. It is not compatible with the standard Atmega2560 bootloader entry in the boards selection menu. Is there any chance this might be the case?

redmuchroom, 2019-08-09:
Hi,

> I can mostly offer suggestions, as I have no hardware available at the moment.

no problem, thanks for your suggestions :)

>If you are using the Ethernet library, it might be getting the IP through DHCP if you have not manually specified it or you are not reading it from EEPROM.

I'm not using the Ethernet lib, only with the eeprom.

> I am assuming you are trying to reset it by ethernet, correct?

indeed but this doesn't work... (reprogram neither...), no reset or restart of the board at all.... it seems that only a power cycle allows to launch the sketch...if I do a reset (on the hardware), the board is stuck in the bootloader...

>Ariadne requires its own board in the board selection menu to upload correctly. It is not compatible with the standard Atmega2560 bootloader entry in the boards selection menu

aah! indeed I was not using it... however, I didn't check this part yet, computer is used by someone else at the moment... 

anyway thanks for the support

redmuchroom, 2019-08-09:
hi,

I got the reset and reprogram command working... thanks to that I can upload with tftp without any problem.
the "local" reset doesn't work but this is fine  for remote flashing :)

just need to figure out why it is not possible to upload with another computer...

thanks for your support!

pillo79, 2019-09-29:
@redmuchroom Hello, I was going to start the same work (add support for the Controllino MEGA). Will you share your current work or create a pull request? Thanks in advance!
Luca

redmuchroom, 2019-09-30:
Hi Luca,

I put this on the side at the moment. In fact with this bootloader you cannot use the monitor as debug ... as soon as you send something on the monitor it will reboot the controllino to be ready for flashing a new sketch...  I didn't spend time yet to trigger this only based on a specific command...

anyway, for flashing the bootloader I used an arduino as described here : http://www.gammon.com.au/forum/?id=11635)
also we need to modify the file spi.c and spi.h as described here : https://github.com/loathingKernel/ariadne-bootloader/issues/28 (remove the SD card pin because it is used by the RTC)
and here https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/issues/13

with this I've been able to load a new sketch remotely. 

let me know if you need more info.