loathingKernel/ariadne-bootloader #36: Download through NAT

GrubSerg, 2019-10-10:
Hello!

There was a problem with the firmware over the network through NAT. The TFTP server generates random ports during data transfer and there is no way to forward them. Is it possible to make a port permanent?

loathingKernel, 2019-10-10:
It is not random. The write request on the arduino is the default 69 and the data transfer port is 46969. If you need to change those, you will need to flash the network settings to the eeprom, or edit and recompile the bootloader

loathingKernel, 2019-10-10:
The communication port is the default 69, and the data transfer port is statically set to 46969. If you want to change them, you will have to edit the source and rebuild the boot loader.

GrubSerg, 2019-10-10:
Thanks for the very prompt reply. Probably the problem is in my port forwarding settings. It doesn’t work out yet. I will think. Did you flash through the router?

loathingKernel, 2019-10-10:
Yes I have flashed through a router from an external server to an Arduino in my LAN.

GrubSerg, 2019-10-10:
I apologize for the persistence, but could not show an example of port forwarding.

loathingKernel, 2019-10-10:
Have you read the documentation [here](https://loathingkernel.github.io/ariadne-bootloader/)?
There is a section, that explains the basics of the process. It doesn't go in detail about how to port forward but has a link to an external wiki for that.

GrubSerg, 2019-10-11:
Thanks for answers. The problem was in the router settings. )))))

closed 2019-10-11 by GrubSerg